meatpiHQ/wican-fw #406: BMW CE02 EV Motorbike

lucabacchipg, 2025-05-08:
Please can you add profile

Soc (log 94%)
Mileage (log 1931km) 

Probabli same profile will be ok also for CE04

[log (2).txt](https://github.com/user-attachments/files/20102081/log.2.txt)

[image]

Thx

meatpiHQ, 2025-05-08:
@lucabacchipg For the Mileage you can do a "Standard PID scan" the Mileage should come up PID 31.

[image]

For the SoC please provide another log, but this time after you connect, in Car Scanner press on the SoC sensor what for about 10 seconds and unplug WiCAN. This way we can isolate which PID is responsible for the SoC.

https://meatpihq.github.io/wican-fw/car_scanner/logs

lucabacchipg, 2025-05-08:
[image]

[log (4).txt](https://github.com/user-attachments/files/20108774/log.4.txt)

This is log. 100% SOC

lucabacchipg, 2025-05-14:
Hello, any news ? My latest log is ok ?

meatpiHQ, 2025-05-14:
Sorry missed your post. The log seems to be missing the configuration part. Can you please redo it?

meatpiHQ, 2025-05-14:
The configuration in the logs look like the screenshot below. These commands are used to initialize the bus controller to read the parameters.

[image]

lucabacchipg, 2025-05-14:
Hello, standard PID works. (6-ISO PROTOCOL) :)

So for now is ok , i don't know if you want to make a profile BMW EV CE02 Motorbyke :)

[image]