meatpiHQ/wican-fw #407: [Feature request] Add 33.333 kbit/s (33k) Baud Rate Support for WiCAN USB

Dicoz, 2025-05-10:
Currently, the WiCAN USB does not support a bitrate of 33k (33.333 kbit/s). This specific bitrate is used in some automotive and industrial applications, particularly in low speed CAN (B-CAN) bus implementations. 

I would like to request the addition of 33.333 kbit/s as a selectable baud rate option in the WiCAN USB configuration settings.

Assuming that a user has access to low speed CAN bus and is able to correctly tap meatPi USB to low speed CAN (B-CAN) bus. It will become possible to listen to all messages flowing through the bus, filter, decode, and send selected values using MQTT transport.

meatpiHQ, 2025-05-13:
@Dicoz Please try this firmware, i've replaced 10K with 33.33K. So for now select 10K and it should give you 33.33K baudrate.

[wican-fw_usb_v412-15-g497ebbe-dirty.zip](https://github.com/user-attachments/files/20191535/wican-fw_usb_v412-15-g497ebbe-dirty.zip)

VladimirKuletski, 2025-05-14:
Thanks. @meatpiHQ 

Downloaded, flashed.
Will check it tomorrow.
Will try to catch any message flow in UI or SavvyCAN.

Any other suggestions?

[image]

[image]

Dicoz, 2025-05-14:
Thanks, @meatpiHQ 
Downloaded, flashed. Will connect it to the car tomorrow.

[image]

[image]

Dicoz, 2025-08-02:
Tried all combinations with "Protocol", "CAN Mode", swapping CAN bus wires - no orange indicator

[image]

Anybody got further?

Dicoz, 2026-09-08:
Finally sorted out.
Was able to gat an oscilloscope and validate the speed of the B-CAN. It's 125 kbit/s.
It started decoding frame as soon as proper speed was setup in the settings.

EOD wasn't able to test 33.333 speed.
So, closing the issue.
Thanks!

closed 2026-09-08 by Dicoz