meatpiHQ/wican-fw #408: VAG MQB ICE custom PID

gamindru, 2025-05-11:
I am trying to configure a custom PID for my car to retrieve the odometer.
In car scanner terminal I am able to extract the odometer value when running the following:
>ATSP6
>ATSH7E0
>2210E0
      7E8037F2278
      7E8076210E000006CB1

In this particular case 6CB1 is the mileage in km.

In WiCAN OBD I have tried to configure both a car profile and a custom PID, however, I am getting no MQTT response.  Standard PIDs work fine.

[image]

Not sure what I am doing wrong. Any help is appreciated.

meatpiHQ, 2025-05-13:
@gamindru in the `initialisation` , `ATSH7E0` needs a `;` at the end.

gamindru, 2025-05-13:
> [@gamindru](https://github.com/gamindru) in the `initialisation` , `ATSH7E0` needs a `;` at the end.

@meatpiHQ that did the trick - I am now getting an MQTT response, albeit with wrong values. Progress!

Referring to my example above, how are bytes greater than B7 referenced in expressions?
Trial and error seem to indicate that the odometer expression should be [B10:B11]

B4_B5_B6_B7_B8_B9_B10_B11
07_62_10_E0_00_00__6C__B1

However, I am getting a value of 0 for [B10:B11]

meatpiHQ, 2025-05-19:
@gamindru The response here has a response pending frame `7E8 03 **7F** 22 78`, 

> B4_B5_B6_B7_B8_B9_B10_B11
> 07_62_10_E0_00_00__6C__B1
> 
> However, I am getting a value of 0 for [B10:B11]

You are correct, but I think you're getting 0 value because you don't always get the response pending frame. Try [B6:B7]

gamindru, 2025-05-19:
@meatpiHQ Thank you for you reply. [B6:B7] returns 4322 decimal - 10E0h is 4320 decimal which seems to confirm the above byte mapping. I have tried various bytes above B7 and all return 0 (as if everything beyond B7 is discarded by WiCAN).

Looking at the Car Scanner logs - most custom PIDs are mode 22 (UDS) and return a pending frame followed by one or more data frames. Is there a reasonably easy way to filter out 7E803**7F**22xx frames (on the WiCAN side)?

If there is no appetite to pursue this please let me know and I will close the issue.

meatpiHQ, 2025-05-20:
@gamindru I would like to solve this issue of course. I think we can have WiCAN ignore the `pending frames`. Are you using the PRO or original WiCAN?

gamindru, 2025-05-20:
@meatpiHQ I am using the original WiCAN.
FWIW:
7E8 seems to be the CAN ID
03 seems to be the number of data bytes in the frame
7F - negative response indicator
22 - mode 22
78  - response pending

gamindru, 2025-05-21:
@meatpiHQ I deleted all settings in Automation and started from scratch - It now works as expected. I am going to assume user error on my part for now. 

For reference, this is what works:

[image]

meatpiHQ, 2025-05-21:
@gamindru please consider creating a profile.