meatpiHQ/wican-fw #411: MQTT Publish Rate

pricard32, 2025-05-13:
I would like to have the option to set a shorter MQTT publish rate for the speedometer.

meatpiHQ, 2025-05-18:
@pricard32 shorter than 1 second ?

pricard32, 2025-05-18:
Yes, something like 500ms should be enough.

My RV use MPH but I need KPH.
I've buy the Meatpi to update my dashboard in KPM but 1sec are just a little too slow to operate correctly.

I understand that if every sensor are updated at the speed it could impact the device's CPU, but if you could allow a couple of PID under 1sec, I would appreciate it !

meatpiHQ, 2025-05-20:
@pricard32 Try adding it as a custom PID.

You might need the init if its the only PID.

[image]