meatpiHQ/wican-fw #413: BLE activation while WifiClient is active

lucabacchipg, 2025-05-15:
I need to have Wican configured as a Wifi Client so that it connects to the network when near an access point.
At the same time I need to always have BLE active for identification.
It is not possible to activate BLE from the menu
Why ? This is bug or need a feature to do ?

FEmser, 2025-05-17:
I need this too. My Wican Pro arrived yesterday and works like a charm, but I need ble for live data connection to abrp and want to send data to home assistant when parked at home. 
What I don't need is the AP mode since I use ble instead anyways.

meatpiHQ, 2025-05-18:
I'm working on that. But most likely it will only be supported on the PRO version.

FEmser, 2025-05-18:
Great to hear that :)
I got the pro version and am looking forward to test this.

meatpiHQ, 2025-05-20:
@lucabacchipg I can send you a firmware with the WiFI\BLE enabled to try, but I'm not about the effects. You plan to use it only as a beacon? 

Why not detect when it connect to WiFi or Home assistant ?

lucabacchipg, 2025-05-22:
> [@lucabacchipg](https://github.com/lucabacchipg) I can send you a firmware with the WiFI\BLE enabled to try, but I'm not about the effects. You plan to use it only as a beacon?
> 
> Why not detect when it connect to WiFi or Home assistant ?

Because when all 4 car are in house, they go in stanby in 15 minuts. So i need to have a BLE active to know presence

meatpiHQ, 2025-05-24:
@lucabacchipg It's not a good idea to have the BLE ON all the time. It will drain your battery.

How about I add a feature for periodic wake up? for example you can set it sleep after 2 min and sleep for 1 hour (or whatever values you set)?