meatpiHQ/wican-fw #415: New EV - Honda Eny1 2025

manaesh, 2025-05-16:
Hi

Just got my Wican pro.

Setup HA Integration and can see voltage etc.

Have used car scanner to connect and get some logs via elm327 (attached)

I tried using the Honda Clarity-2018 with no luck (no sensors created.

Can a profile be created please?

Many thanks!!!

Mike

manaesh, 2025-05-17:
[log.txt](https://github.com/user-attachments/files/20265328/log.txt)

manaesh, 2025-05-18:
Have realised the logs are rubbish.

I will relog required sensors and upload the logs!

Thanks for a great product! Only way to get some data from the Honda seeing as there is no API access to MyHonda+ yet! :(

Thanks

Mike

meatpiHQ, 2025-05-18:
@manaesh please follow the instructions in link to get the logs for a specific PID 

https://meatpihq.github.io/wican-fw/car_scanner/logs

manaesh, 2025-05-19:
Hi @meatpiHQ 

Using that profile (And I agree the PIDs match!) - I get 0 from both /autopid_data in in home assistant : 

{"SOC":0,"ODOMETER":4214}

Any hints on what I am doing wrong?

Thanks!

Mike

meatpiHQ, 2025-05-19:
@manaesh There is a bug, I'll send you a firmware update to fix it. These Hondas are so special I have a dedicated profile in my simulator just for them :)

manaesh, 2025-05-19:
Thanks @meatpiHQ - let me know how to get the firmware when ready?

Happy to help test anything needed!

Thanks

Mike

manaesh, 2025-05-20:
Thanks @meatpiHQ !

That FW sorted it - now getting the SOC though!

Be interesting if anyone else can find any more usable data from the Honda!

Thanks for a great product!

Cheers

Mike

closed 2025-05-22 by manaesh