meatpiHQ/wican-fw #417: New Vehicle - Toyota Proace City Verso 2019-2025

thecodingmax, 2025-05-16:
Hello,
just wanted to ask if it's possible to add the Toyota Proace City Verso 2019-2025 to the supported vehicle list. I have just received my WiCAN Pro and would love to fully integrate it. 
Best regards
Max

meatpiHQ, 2025-05-18:
@thecodingmax Please follow the instructions to export the logs for the SoC and other PIDs you want 

https://meatpihq.github.io/wican-fw/car_scanner/logs

tqma1, 2025-05-25:
This car is also build by Stellantis.
So you can try Fiat and Peugeot car profiles.
May they fit also for the Proace Verso.

thecodingmax, 2025-05-26:
Thank you very much for this hint, I will give it a try.
I am just missing a sensor for the fuel tank, maybe i will find it through one of the profiles of other Stellantis brands ;-)