meatpiHQ/wican-fw #418: LeafSpy support

danjulio, 2025-05-17:
Is it possible to use WiCAN PRO with LeafSpy Pro?  It doesn't seem that LeafSpy Pro sees the WiCAN using the Bluetooth interface.   I have connected using the WiFi interface and pulled some data, however it seems (and I could be wrong) that LeafSpy hangs up trying to send some ELM327 commands (not sure which, trying to figure out how to get a log out of LeafSpy.

danjulio, 2025-10-17:
@meatpiHQ I just tried with V4.40 and unfortunately that does not fix the problem.  Now v2.0.0(10) en IOS LeafSpy Pro crashes after a few seconds of connecting.  I haven't been able to try to get a log file out of it to see if there is any useful information there but I'll keep trying.

meatpiHQ, 2025-10-21:
@danjulio I had a quick look and it seems like LeafSpy does not support some other adapters as well.

I think we can enable the USB serial log output to see what is going on. 

Plug in the USB cable and connect putty set it to serial, baudrate 2000000. To enable the debug mode you need to boot in safemode. To do that all you need to do is hold the button and while pressing the button insert into the OBD port, and keep pressing for 8 seconds or so. Then press `Enter`, you should get a command prompt `wican> ` use the command `debug -e 1` and then `system -r` to reboot.

Save the logs while you try to connect using LeadSpy.