meatpiHQ/wican-fw #424: Vehicle Profile 2005 Dodge (Mercedes) Sprinter

sskerry, 2025-05-20:
Hello

Hoping for help with this one, tried to figure it out on my own but not getting anywhere...

Connected via car scanner works, shows: ISO 14230-4 KWP fast baud init 10.4 kbaud

Here is the car scanner log

===== LOG STARTED: 20.05.2025 16:34:25=====
=====Version=1.118.0 Build=1.118.0 P=False NoDelayELM327Init=False
====Connection profile=Mercedes-Benz OBD-II / EOBD
[Connection type: WIFI]

[WiFi name: <null>]

[Connecting to 10.5.20.191:35000]
client = new TcpClient();
BEFORE await client.ConnectAsync(server, port);
client.Connected
got stream from client

[Connected to 10.5.20.191:35000]
ATZ
ATE0

[ELM RESET]
ATZ

ELM327 v2.3

>ATE0
OK

>?

>?

>?

>?

>?

>?

>?

>?

>?

>?

>
Initialize(initMode=Default)

InitializeDefaultInitString(ProtocolNumber=0,initMode=Default,useATST96=False)
ATE0
OK

>ATE0
OK

>STI
STN2120 v5.8.1

>
[STSupported=True(ByResponse)]VTI
MIC3624 v2.3.18

[... 3947 lines cut ...]
WiFiPort=35000
WiFiServer=10.5.20.191
ZeroConsumptionPIDId=-1
ZeroConsumptionWhenZero=False

Override_roles=
ATST=64
P=False
FPGC=10
CC=0

OS version=18.4.1
Platform=iOS
Idiom=Phone
DeviceModel=iPhone15,2
DeviceManufacturer=Apple

OrigCulture=en-CA
OrigUICulture=en-CA
RunningOnMac=False
Version=1.118.0/1.118.0

PrVersion=1.118.6
#iOS
[***/// END OF CONTACT DEVELOPER REPORT]

sskerry, 2025-05-21:
I should note, this is the wican pro, and I connected to the van via can scanner -> wifi -> wican pro

meatpiHQ, 2025-05-22:
@sskerry Thanks! this is the first KWP vehicle! I need to update the firmware to support it.

I will send you a new firmware version to test in a couple of days.

meatpiHQ, 2025-06-01:
@sskerry Here you go. All you need to do is set the standard protocol to Automatic and try to scan. One thing to note in this new firmware sleep mode will automatically be enabled so you might want to disable it while your testing and enable afterwards, when you're done.

Let me know if you find any issues.

[wican-fw_obd_pro_v422p-beta.zip](https://github.com/user-attachments/files/20540415/wican-fw_obd_pro_v422p-beta.zip)

meatpiHQ, 2025-06-02:
@sskerry Clearing DTC using auto pid or mqtt is not yet supported. Please check in a couple of weeks I plan to look into that.

rawanamoodi-debug, 2026-02-17:
@meatpiHQ How do I know if my version includes fault detection DTC ?