meatpiHQ/wican-fw #435: New Vehicle: Toyota Tundra

joelishness, 2025-05-29:
Hello,
I am opening this issue per the Documentation for Home Assistant:

7. Enable Vehicle Specific Settings:
On the webpage, go to the Automate tab and enable Vehicle Specific settings.

8. Sync Vehicle Model:
Press the ⟳ button next to Vehicle Model to fetch supported vehicle models.

9. Select Vehicle Model:
Select your vehicle model from the dropdown list.
If your vehicle model is not yet supported, please open an issue on GitHub, and we will walk you through adding support.

Thanks!

meatpiHQ, 2025-06-02:
@joelishness Are you using WiCAN or WiCAN PRO?

Have you tried Standard PID scan?

meatpiHQ, 2025-06-03:
@joelishness Try to reinstall the HA integration. Also try putting this in your browser see if it returns the values

http://device_ip/autopid_data

Please update to the latest firmware version.

https://github.com/meatpiHQ/wican-fw/releases/tag/v4.22p-beta

joelishness, 2025-06-07:
**Update**
I finally had time to return to this today, and the two arbitrary entities I had added before now show up!

I added more entities but they didn't show up again, even after the WiCAN rebooted.

Then I restarted Home Assistant = all the entities show up!

This is good, I'm happy with to be making progress.

**Suggestions**
So far I see two points of improvement for the documentation:
1. Start with Standard PIDs (before going straight to Vehicle Specific)
2. Restart Home Assistant so it will pick up new Sensors from the added PIDs

**Replies**

> Try to reinstall the HA integration.

I have not done this, doesn't seem necessary.

> Also try putting this in your browser see if it returns the values
> http://device_ip/autopid_data

Yes, this returns a json page with all the PIDs that were added under Automate.

> Please update to the latest firmware version.
> https://github.com/meatpiHQ/wican-fw/releases/tag/v4.22p-beta

I am currently on latest stable (v4.11p).
Should I still do this?

**Going forward**
What is the next step, please?

Thanks so much!

joelishness, 2025-06-19:
> [@joelishness](https://github.com/joelishness)
> 
> > Suggestions
> > So far I see two points of improvement for the documentation:
> > Start with Standard PIDs (before going straight to Vehicle Specific)
> > Restart Home Assistant so it will pick up new Sensors from the added PIDs
> 
> Will do.
Thanks very much!
Please let me know if you would like assistance with this, I could take a stab at it if that would be needed or helpful.

> > What is the next step, please?
> 
> Is there any parameters you want that are not available in the standard PIDs?
I would love to get Odometer and Fuel tank level readings.
Is it possible to see if these are available?

And if it were possible to log any warning indicators (tire pressure, check engine, etc) that would be really great.

meatpiHQ, 2025-07-27:
@joelishness Please follow the instructions to export the logs, we have to do each sensor on its own lets start with the Odometer.

https://meatpihq.github.io/wican-fw/car_scanner/logs