meatpiHQ/wican-fw #437: WiCAN Pro + Busmaster "Failed to start the hardware"

Schmirgus, 2025-05-29:
Hello everyone :),

it seems i am doing something wrong. I cant figure out how to connect the WiCAN Pro to Busmaster. I followed [this guide](https://meatpihq.github.io/wican-fw/busmaster/usage) and downloaded the linked Busmaster version but i think i am missing something since i am getting the error "Failed to start the hardware" in Busmaster. Currently i am on FW 4.11 but i will update it now to 4.12 and try it again.

In general the WiCAN Pro should work with Busmaster right?

[image]
[image]

thanks in advance <3

Schmirgus, 2025-05-29:
So i tried again to get it working. Everytime i press the connect button it seems that the WiCAN is rebooting. The blue LED is constantly on before  a press the connect button and  when i press the button the blue LED turns off until the WiCAN reboots.

https://github.com/user-attachments/assets/a6005be0-142a-49d6-a980-d1b9a93274ce

meatpiHQ, 2025-06-01:
@Schmirgus Thanks for reporting this. The issue was addressed in this release. 

Please let me know if you found any other issues.

https://github.com/meatpiHQ/wican-fw/releases/tag/v4.22p-beta

Schmirgus, 2025-06-01:
Thank you for the update :) i will give a feedback when i tried it out. Just flashed the beta release and at least on my side it _disabled_ sleepmode instead of enabling it.

Greetings,
Mirgus