meatpiHQ/wican-fw #460: New Vehicle Request:  2025 Toyota Tundra

tgszo, 2025-06-25:
Requesting as per the instructions on the web site.

meatpiHQ, 2025-06-29:
@tgszo please try scanning for standard parameters first

https://meatpihq.github.io/wican-fw/config/automate/usage#standard-pids

CarlsonCustoms, 2025-09-14:
@meatpiHQ , can you also provide the instructions for actually getting a new vehicle requested? I have a 2019 Toyota Tundra and did the standard scan and the specific pids I am looking for (Fuel Tank Level, Odometer, and Transmission Pan Temperatures) do not show up in the standard scan. When I use the car scanner app (which I found out the frustrating way that you can't have the Wicam in AP+base station mode for it to connect) I am able to see these PIDs without issue. I used the instructions to export the logs but I'm not understanding the next step to actually create the vehicle profile or request one with the data provided from the logs.