meatpiHQ/wican-fw #463: New Vehicle Request: 2016 Opel Astra K

Juliku, 2025-07-03:
Standard PIDs got me some values but these are missing and would come quiet handy:

- Odometer
- Tire pressures
- Oil remaining lifetime
- Average Fuel Consumption

These values are visible with carscanner.

What logs/data can i collect to identify the right settings?

Thanks in advance

meatpiHQ, 2025-07-27:
Thanks I will create the profile soon

Juliku, 2025-08-11:
Wow, thank you, nice work.
Only the odometer seems a few kilometers off, but it is that way in carscanner, too.
I'll gather a few datapoints, maybe the factor is not quiet right.

OTTO-KuchlbauerJ, 2025-08-28:
I found out dividing the raw value just by 64 comes the closest (2km deviation) and sounds very reasonable.