meatpiHQ/wican-fw #468: No CAN data 2023 Outlander PHEV

amwalters, 2025-07-06:
I'm probably missing something simple here, but I cannot get any CAN data to display from my WiCAN-OBD-PRO running v4.2.3.

Successes:  
- I can get basic PID data with CarScanner [ISO 15765-4 CAN (29 bit ID, 500 Kbaud)]
- I can connect to a terminal and interact in ELM327 mode using: 
`socat -d -d -lf /tmp/socat.log STDIO,echo=1 TCP:[192.168.88.30:35000](http://192.168.88.30:35000/),crnl`

Failures:
- ATMA fails to produce any output in my ELM terminal
- No data shows up from any baud rates in the WiCAN monitor page
- WiCAN is found by SavvyCAN, but there is no data. Just repeated:
```
Write to serial -> f1 9 
Got data from serial. Len = 4
f1 09 de ad
```
- No data seen by candump in slcan mode.

I've tried every port and baud rate combination, but no luck. Any way to troubleshoot this?

amwalters, 2025-07-10:
SOC is not available in CarScanner, and I have not found anyone who has figured out how to retrieve it since 2023 models were released. My hope was to sniff it out.

amwalters, 2025-07-20:
I've made some progress and have a working SOC in carscanner, but can't get it working with AutoPID. I'm using PID 2201D1 with an equation of B*0.64-19.2.

CarScanner Logs [here](https://drive.google.com/file/d/1HTwaCv13-dSpNg61EsuKEpOs23xTcx_l/view?usp=drive_link)

I can't figure out how to do this in carscanner, but adding 'ATCRA18DAF1DB' filters to only the BMU, so you don't get the negative responses seen when carscanner is polling.  You can also seeing carscanner sends a header, 18DB33F1, that I was trying to use to filter those other responses out (clearly not working).

Something that would really help troubleshooting custom PIDs in WiCAC, which is available in CarScanner, would be to add a test button under the custom PID's so you could see the response.  If I could see the exact response I was getting for each PID, I think I could get it figured out on my own.

amwalters, 2025-07-23:
Profile works! I'm now manually plugging through these PID codes to see what all I can pull out. I'll get back to you with an updated profile when I have something usable.

miroslavvuj, 2025-08-16:
@amwalters Great stuff! I'd be very keen to find out about your progress as I too have an 2023 Outlander PHEV. I am expecting the delivery of my WiCAN Pro any day now and would be happy to get involved in improving results.

meatpiHQ, 2026-04-02:
@wambs the issues with sleep you're having is related to the changes you made on your fork, its not in main WiCAN PRO firmware.