meatpiHQ/wican-fw #471: Autopid Error Topic is missin

panzerdev, 2025-07-09:
Hi! I just purchased a few MeatPi devices and had issues setting it up. After trying out a few Firmware version I saw that the `error_topic` from the last 3.x fw version is gone in 4.12. 

Why is that and how can I debug over mqtt if something goes wrong with the can bus calls.

meatpiHQ, 2025-07-10:
@panzerdev I can bring it back, but it's not useful for most users. I suggest the you connect using a terminal over TCP to port 3333 and use ELM327 command directly  once you figure out the correct command sequence then fill in the autopid.

meatpiHQ, 2025-07-13:
@panzerdev I think we should try and fix the issues with the profile. Which profile are you using?

I will add the debug topic in the next revision.

panzerdev, 2025-07-13:
> I will add the debug topic in the next revision.

Thant sounds great!

> Which profile are you using?

I have an Ioniq 5 and I would've done an own issue for my issues with that profile, but I wanted to debug it first. The value I saw errors for was the `ODOMETER` but I had no time to further investiage and I will need to make myself a bit more familiar with the whole CAN ELM327 protocol.