meatpiHQ/wican-fw #473: Add support for Zeekr 7X

ff20000, 2025-07-10:
I like to get SoC info from the car. To use in ABRP.

meatpiHQ, 2025-07-17:
Have you tried setting the protocol to ELM327 and connect using ABRP?

ff20000, 2025-08-02:
Is this possible with the non pro version?

meatpiHQ, 2025-08-04:
unlikely, chatgpt only knows the data it's trained on, this info is not yet out in the public

ff20000, 2025-08-25:
Yes, connecting with home assistant is my goal. I tried to connect my wican to savvycan but i didn't get it working yet. No info is comming out of my Zeekr.

jandetlefsen, 2025-10-28:
That didn't happen. Now the dealer told me mid November.

ff20000, 2025-12-16:
And are you driving a zeekr 7x yet?

jandetlefsen, 2026-01-02:
@ff20000 on it now, but need to get an SD card first to use the logger. Have you tried any of the similar cars like volvo or zeekr?

jandetlefsen, 2026-01-06:
Having a hard time connecting to the ECU with Car Scanner Pro and getting any non zero values out of it.

rob0101, 2026-01-11:
> Having a hard time connecting to the ECU with Car Scanner Pro and getting any non zero values out of it.

same here.  Think I saw the VIN number and that's it.

jandetlefsen, 2026-01-11:
I've read somewhere that the OBD access for normal users might be encrypted and workshops using online services for read out. Is that a thing in the automotive space?

meatpiHQ, 2026-01-11:
@jandetlefsen are you using WiCAN or WiCAN PRO?

jandetlefsen, 2026-01-11:
WiCan Pro

jandetlefsen, 2026-01-11:
This is what happens when i select the zeekr profile

`Stopped

===== DISCONNECTED AT: 11.01.2026 11:22:37
[///**** CONTACT DEVELOPER REPORT:[Car Scanner settings:]
AccelerationItems=5
AccelerationTestSpeedSourcePid=13
AccelerationUseG=True
AdaptiveTimings=1
AddNissanConsult3Pids=False
AllowProfilesBackgroundUpdate=True
AlwaysPingECU=True
AlwaysRecordFuelConsumption=False
ATCommandStateOptimization=True
ATCRAOptimization=False
ATSTIdx=6
AutomaticallySwitchTheme=True
AutomaticallySwitchThemeAvailable=True
AutomaticReoptimization=True
BindSocketToFirstWiFiNetwork=True
BindSocketToWiFiNetwork=True
BluetoothLocationWarningShowed=False
BoostCalculationMethod=Auto
BoostCalculationMethodIdx=0
BrandForDTC=Volvo
BTDeviceID=
BTDeviceName=
BTLEDeviceID=
BTLEDeviceName=
BTLEInputID=
BTLEOutputID=
BTLEServiceID=
BTLEShowDevicesWithoutName=False
CalibrationStartOdometer=0
CalibrationStartTime=1/1/0001 12:00:00 AM
CANOptimizationWarningShowed=True
CANOptimizeMaxInRequest=6
CANOptimizeMode22=True
CANOptimizeMode22MaxInRequest=3
CANOptimizeMode22SelfLearningMode=True
CANOptimizeRequests=False
CANRequestSegmentationSTNLevel=True
CANResponseSegmentationSTNLevel=False
CarPlayDashboardSelectedIndex=0
CarPlayUpdateInterval=300
ChartDisplayStyle=FastLine
ChartShowAverageValue=False
ChartsView=0
ChartsVisible=2
CheckBluetoothTurnedOn=True
CheckIsELMWhenConnecting=False
CheckOnlyPositiveResponseMarker=False
CodingLastPlatformSelected=
ConnectionType=WiFi
ConnectOnLaunch=False
Currency=$
CurrentCarId=638890611869698600
CurrentCarName=My car
CustomFuelAF=14.7
CustomFuelDensity=0.73
CustomInitString=ATZ
ATE0
ATH1
ATSP7
ATS0
ATM0
ATAT1
ATSHD01635
ATCP1D
[... 414 lines cut ...]
WiFiPort=23
WiFiServer=192.168.0.133
ZeroConsumptionPIDId=-1
ZeroConsumptionWhenZero=False

Override_roles=
ATST=64
P=True
FPGC=26
CC=0

OS version=26.1
Platform=iOS
Idiom=Phone
DeviceModel=iPhone17,1
DeviceManufacturer=Apple

OrigCulture=en-US
OrigUICulture=en-US
RunningOnMac=False
Version=1.121.0/1.121.0

PrVersion=2.0.60
#iOS
[***/// END OF CONTACT DEVELOPER REPORT]`

meatpiHQ, 2026-01-11:
@jandetlefsen please export the logs and post them here. Step 8 in the link below

https://meatpihq.github.io/wican-fw/car_scanner/logs

jandetlefsen, 2026-01-11:
[log.txt](https://github.com/user-attachments/files/24548128/log.txt)

meatpiHQ, 2026-01-11:
@jandetlefsen I think car scanner have not yet found the correct PIDs for this model. Please try other apps like Torque... you might find one that works.

jandetlefsen, 2026-01-11:
ok

jandetlefsen, 2026-01-12:
okay i've treid those apps with no luck

- AutoDoctor
- Car Scanner 
- Carista OBD2
- Carly - OBD2 Car Scanner
- Garage Pro
- Infocar - OBD2 ELM Scanner
- iOBD2
- OBD2 Scanner: Torque ELM 327
- OBDLink
- OBD2 Car Scanner 
- OnTrack - OBD2 Scanner 
- Torque OBD Lite - Car Scanner 
- Torque OBD2
- Torque Pro: OBD2 Car Scanner