meatpiHQ/wican-fw #478: 28kWh Ioniq - Odometer

dhi20, 2025-07-16:
Has anyone been able to get the odometer to work for the 28kWh Ioniq?

I found the following reference to it from a torque spreadsheet - 
000_Odometer | Odo | 22b002 | Int24(g:h:i) | 0 | 16777215 | km | 7c6

But I can't work out how to do this with WiCan

I have managed to get SOC_D using PID 21057 which appears in the torque sheet as  
000_State of Charge   Display | SOC Display | 2105 | af/2 | 0 | 100 | %

meatpiHQ, 2025-07-17:
@dhi20 have you tried car scanner app? does it show there? If it does please export the logs 

https://meatpihq.github.io/wican-fw/car_scanner/logs

dhi20, 2025-07-17:
@meatpiHQ Thanks, the car scanner app shows odometer = 0. I've been able to access it via OVMS and ESPHome

- code from ovms and esphome -     
- 'cmd': b22b002,
    'canrx': 0x7ce,
    'cantx': 0x7c6,
    'optional': True,
    'fields': (
        {'padding': 9},
        {'name': 'odo', 'width': 3},
        {'padding': 3},

meatpiHQ, 2025-07-20:
@dhi20 I've update the Ioniq 2017 profile please try it and let me know if the odometer is correct.

jonasblomq, 2025-07-21:
Tested the new profile on a 38kWh ioniq 2020, did not work...

meatpiHQ, 2025-07-27:
@jonasblomq What about the other Hyundai Ioniq profile did you try it?

philipkocanda, 2025-08-13:
This works for mine (Ioniq 28 kWh, 2017):

```
{
  "pid": "22B002",
  "pid_init": "ATSH7C6;ATFCSH7C6",
  "parameters": [
    {
      "name": "ODOMETER",
      "expression": "[B12:B14]",
      "unit": "km",
      "class": "distance",
      "period": "5000",
      "min": "0",
      "max": "",
      "type": "Default",
      "send_to": ""
    }
  ]
},
```

The cause of the problem is that the PID IDs do not map 1:1 with the actual UDS payload. UDS Payload:

```
62 B0 02 E0 00 00 00 00 AB 00 FE 07 00 00 00
                           ^^ ^^ ^^
```

timwelchnz, 2025-10-16:
Will this be the same for https://github.com/meatpiHQ/wican-fw/issues/306 ??