meatpiHQ/wican-fw #479: WiCAN (standard) possibly triggering ABS fault?

elidhu, 2025-07-17:
My vehicle (2014 Toyota FJ Cruiser) is intermittently triggering a fault with the ABS system (steering angle sensor). It seems to only occur when the WiCAN is plugged in, however it is intermittent!

It is reminiscent of this => https://github.com/meatpiHQ/wican-fw/issues/397 although in my case it is just a single fault.

[config_2025-07-15.json](https://github.com/user-attachments/files/21277221/config_2025-07-15.json) 

Is it possible that the WiCAN is causing this? how can I debug it?

meatpiHQ, 2025-07-17:
@elidhu Please try this firmware version and let me know if it resolves the issue.

[wican-fw_obd_v412-63-gc8d257c.zip](https://github.com/user-attachments/files/21300616/wican-fw_obd_v412-63-gc8d257c.zip)

elidhu, 2025-07-19:
I can't say for _certain_ that it has, but it seems it may have...

Were there any relevant changes? or was this just an "upgrade and see"? (just out of curiosity)

meatpiHQ, 2025-07-19:
There was an issue with Autopid process that was fixed in the update. This issue does not seem to affect most vehicles but possibly can affect some, so it's also upgrade and see.

elidhu, 2025-07-19:
Easy, thank you for the info!

closed 2025-07-19 by elidhu