meatpiHQ/wican-fw #487: BYD Atto 3 not working with WiCan Pro on 4.23p

kirashogun, 2025-07-28:
Just got my WiCan Pro today. Plugged it in, configured AP, switched to Auto PID. At this stage, everything is fine except all the reading I can get is the 12V battery status. So far so good.

If I goto Automate, and load Vehicle Specific PIDs (Atto 3), when I save settings, it would reboot and seem to boot back up again but immediately crashes, with blue LED slowly blinking, no Wifi connectivity at all, dead device. I have to erase config and reflash it from USB to wake the device up.

Can someone help me sort out what is going on here? Right now, the WiCan Pro is just a expensive doorstop.

meatpiHQ, 2025-07-28:
@kirashogun After you erase and reflash the firmware you can connect to it's AP is that correct?

meatpiHQ, 2025-07-28:
Can you log the serial output? Use putty or teraterm baudrate 2000000, Serial-A. This will tell us exactly what's going on

meatpiHQ, 2025-07-28:
Thanks I'm try to reproduce here

kirashogun, 2025-07-28:
No worries, I'm also based in Melbourne so if you really need, we can meet up and help you debug this issue.

dral3x, 2026-01-28:
Offtopic

@kirashogun I have a BYD Atto 3 as well, and I'm looking for confirmation of which information you are getting from the WiCan Pro. I want to track them on Home Assistant while the car is at home. Thanks!