meatpiHQ/wican-fw #491: VW Polo 2022

hhielkema, 2025-07-29:
G' Day.

I am trying to add my "VW Polo 2022 (regular gas)".
"
Using WiCAN with Home Assistant
9. Select Vehicle Model:
Select your vehicle model from the dropdown list.
If your vehicle model is not yet supported, please open an issue on GitHub, and we will walk you through adding support.
"
But is not available.
I only see the electric vehicles.
Is it possible to add this make and model.

Kind regards 
Hans

meatpiHQ, 2025-08-03:
Hey @hhielkema please try to scan for standard pids first, if there's any car specific parameters then we can create a profile

https://meatpihq.github.io/wican-fw/config/automate/usage#standard-pids

hhielkema, 2025-08-08:
Hi.
I have run the scan for the standard-pids.
The list, if i am correct, is:

01-MonitorStatus
03-FuelSystemStatus
04-CalcEngineLoad
05-EngineCoolantTemp
06-ShortFuelTrimBank1
07-LongFuelTrimBank1
0B-IntakeManiAbsPress
0C-EngineRPM
0D-VehicleSpeed
0E-TimingAdvance
0F-IntakeAirTemperature
11-ThrottlePosition
13-OxySensorsPresent_2Banks
15-OxySensor2_Volt
15-OxySensor2_STFT
1C-OBDStandard
1F-TimeSinceEngStart
20-PIDsSupported_21_40
21-DistanceMILOn
2E-CmdEvapPurge
2F-FuelTankLevel
30-WarmUpsSinceCodeClear
31-DistanceSinceCodeClear
33-AbsBaroPres
34-OxySensor1_FAER
34-OxySensor1_Crnt
3C-CatTempBank1Sens1
40-PIDsSupported_41_60
41-MonStatusDriveCycle
42-ControlModuleVolt
43-AbsLoadValue
44-FuelAirCmdEquiv
45-RelThrottlePos
46-AmbientAirTemp
47-AbsThrottlePosB
49-AbsThrottlePosD
4A-AbsThrottlePosE
4C-CmdThrottleAct
51-FuelType
53-AbsEvapSysVapPres
56-LongSecOxyTrimBank1
56-LongSecOxyTrimBank3
5E-EngineFuelRate
60-PIDsSupported_61_80
62-ActualEngTorqPct
63-EngRefTorq
65-AuxInputOutput
67-EngineCoolantTemp1
67-EngineCoolantTemp2
6D-FuelPresContrSys
70-BoostPresCntrl
71-VariableGeoTurboVGTCtr
73-ExhaustPressure
77-ChargeAirCoolerTemperature
80-PIDsSupported_81_A0
8B-DieselAftertreatment
8E-EngineFrictionPercentTorque
9D-EngineFuelRate
9E-EngineExhaustFlowRate
A0-PIDsSupported_A1_C0
A4-TransmissionActualGear

Is this what you mean.

Kind regards.
Hans