meatpiHQ/wican-fw #497: New Vehicle Request: GMC Sierra

daaronbramble, 2025-07-30:
Hi there, I've installed WiCAN on my truck but don't see it in the vehicle list. It is a 2020 GMC Sierra 1500 AT4. Please let me know what to do to get it working. Thanks!

- Aaron

Ken7382, 2025-08-01:
Get in the truck, start the engine and while it's running, go to the "automate" page and hit "scan PID's" under the Standard PIDs section. That will show you the available PID's you can read, and add them. (The dropdown box next to the button will populate after a few seconds.) I have my 2015 Tahoe on it just fine but there just isn't much info coming out by default; for that, you'll need additional PID's (which I don't know yet.) My 2017 Impala is the same way.

[image]

What are you using it with? I'm using mine with Home Assistant to show me my fuel tank levels so I know which vehicle I can take in a hurry (not have to fill up) if I'm in a rush.

[image]

I just got mine a few days ago too, so far VERY cool but right now I'm not able to see some of the things I want (like oil life, tire pressure sensors) because they probably require some sort of code to send so that the ECM/BCM will send back that data.

Ken7382, 2025-08-02:
Awesome, glad it worked! It's a bit to work out but once you figure it out, it works great. The HA integration works great too. Just don't be a bonehead like me and think you have to write your own MQTT sensors - all that is done for you in the HA integration. 

I think to get the oil life and tire pressure you have to send a value to the ECU - otherwise you just get the standard ones. I saw on a post about SS Camaros that someone figured out the tire pressure one but I haven't dug much beyond that. Not sure how to implement it either.

Also, make sure your power saving feature is set at a reasonable level, otherwise the device resets and sends a static value for the fuel level (92.55%) for some reason. That'll make it look like you have plenty of gas, when you don't! 

Be sure to give your device a static IP (or reserve one) in your router too - if the IP changes it will stop working. Found that out yesterday when I came home from a shopping trip.

Ken7382, 2025-08-04:
OK, for other PID's I found some docs for you to peruse:

[Document for ELM327 device](https://cdn.sparkfun.com/assets/learn_tutorials/8/3/ELM327DS.pdf)
This has the AT codes in it (next file shows how they're used in a config file) - it's not what WiCAN uses of course, but it shows the commands generic to OBD2 itself.

[JSON formatted config file for Chevy Bolt](https://github.com/meatpiHQ/wican-fw/blob/main/vehicle_profiles/chevrolet/bolt.json)
This has configuration for a Chevy Bolt - not relevant for either of us, BUT it shows how the codes need to be formatted.. 

So let's look at this code inside the Bolt file:

`  "car_model": "Chevrolet: Bolt",
  "init": "ATSP6;ATST96;",
  "pids": [
    {
      "pid": "2283341",
      "pid_init": "ATSH7E4",
      "parameters": {
        "SOC": "(B4*100)/255"`

From what I can tell:

- Model is obvious, it's more for us to select than a program to need (human friendly name)
- Init values are ATSP6 and ATST96
  - ATSP6 = set protocol to 6 (no idea what that is)
  - ATST96 = set timeout to 96 (multiplied by 4 milliseconds, so 384 milliseconds)
- PID 2282241 is a PID for state of charge (SOC) that's unique to the Bolt or at least Chevy EV's
- PID INIT = ATSH7E4 - this sets the header to 7E4 (no idea what the header is)
- parameters just defines what the raw number should be transformed into for it to be a real value (keep in mind that some A/D converters for analog values are 8 bit and only spit out a value of 0-255)

[In this post on a Camaro SS forum..](https://www.ssforums.com/threads/obd2-pid-codes.41346/?post_id=913642&nested_view=1&sortby=oldest#post-913642)
They show some PID's for tire pressure and temperature.

Haven't found anything else like oil life (which is a big one for me since I of course do my own oil changes.) Odometer would be nice too but not a hard requirement for me.