meatpiHQ/wican-fw #503: New Vehicle Audi A4 B8 (2011) - DPF measurements, oil temperature and fuel level

adriannow, 2025-08-03:
Got a WiCan Pro last week and started from scanning standard PIDs - a lot of them work out of the box, which is great. I managed to correctly implement also two additional parameters related to DPF filter:

`    "pids": [
      {
        "pid": "22114F",
        "pid_init": "ATSH7E0;",
        "parameters": {
          "DPF_SOOT_CALC": "[B4:B5]/100"
        }
      },
      {
        "pid": "22114E",
        "pid_init": "ATSH7E0;",
        "parameters": {
          "DPF_SOOT_MEAS": "[B4:B5]/100"
        }
      },
    ],`

Where I failed however, was in implementing oil temperature and fuel level. I ended up recording logs from CarScanner, but I'm having hard time understanding them. 

For Oil temperature (from around 25-30 degrees Celcius - the car was just starting to heat up):
[oil-temperature-carscanner-log.txt](https://github.com/user-attachments/files/21566704/oil-temperature-carscanner-log.txt)

And for fuel level, there were actually 2 values: fuel level (28 L) and free space in fuel tank (around 35 liters) - any of them working would be awesome.

[free-space-in-fuel-tank.txt](https://github.com/user-attachments/files/21566723/free-space-in-fuel-tank.txt)
[fuel-level.txt](https://github.com/user-attachments/files/21566722/fuel-level.txt)

I'd be grateful for help in implementing those - or pointing me to the resources where I could learn how to decode the CarScanner logs - there's so much more sensors I'd be happy to integrate and share here with others to use.

adriannow, 2025-08-04:
@meatpiHQ I did just that - searched for all sensors, selected the interesting one - let it record for about 30 seconds, then unplugged the dongle, disconnected in the app and then exported logs (maybe for the oil temperature it took to long before I've found the correct one on the list. 

[image]
[image]

I've tried to figure out something from the logs after importing them to SaavyCan (had to trim out just to have end of the file, otherwise it was crashing), but I'd need more experience with this to came up with something useful.

If you still think the logs are incorrect, I'm not sure what I could do differently - but I can redo them if you think that would help.

adriannow, 2025-08-04:
I've prepared the logs again:

[fuel-level2.txt](https://github.com/user-attachments/files/21583935/fuel-level2.txt)
[oil-temp2.txt](https://github.com/user-attachments/files/21583936/oil-temp2.txt)

[image]
[image]

adriannow, 2025-08-05:
For the DPF related PIDs - those seem to function correctly:

Name | Init | PID | Expression | Min Value | Max Value | Unit
-- | -- | -- | -- | -- | -- | --
DPF_SOOT_CALC | ATSH7E0 | 22114f | [B4:B5]/100 | 0 | 40 | gr
DPF_FULLNESS | ATSH7E0 | 22114f | [B4:B5]/24 | 0 | 100 | %
OIL_ASH_RESIDUE | ATSH7E0 | 22178c | [B4:B5]/10 | 0 | 80 | gr
DPF_SOOT_MEAS | ATSH7E0 | 22114e | [B4:B5]/100 | 0 | 40 | gr
DIST_LAST_REGEN | ATSH7E0 | 221156 | [B6:B7]/1000 | 0 | 1000 | km
TIME_LAST_REGEN | ATSH7E0 | 22115e | [B6:B7]/60 | 0 | 2000 | min
DPF_IN_TEMP | ATSH7E0 | 2211b2 | ([B4:B5]-2731)/10 | -40 | 1000 | °C
DPF_OUT_TEMP | ATSH7E0 | 2210f9 | ([B4:B5]-2731)/10 | -40 | 1000 | °C

adriannow, 2025-08-12:
@meatpiHQ I'm having trouble with creating `expression` in WiCan interface for odometer. I have my custom PID that correctly fetches the data, but it's not parsed properly i think.

All i can see is the raw frame in the MQTT (but not separate parsed value):

`wican/9888e006736d/can/rx {"bus":"0","type":"rx","ts":41827,"frame":[{"id":2024,"dlc":8,"rtr":false,"extd":false,"data":[7,98,22,169,0,4,222,194]}]}`

Using this formula I've correctly decoded this frame to 319170 km.
`    5801: ["Odometer", "km", "mdi:counter", lambda a, b, c, d: (a * 16777216) + (b * 65536) + (c * 256) + d],
`
Could you help with preparing correct expression that would handle extracting correct value from this frame by WiCan Pro itself?

meatpiHQ, 2025-08-12:
@adriannow Great work! Did you setup the Destination Type and topic in the custom PID?

It would be awesome if you can create a profile, if not I can make one based on the data you provided.

adriannow, 2025-11-17:
@meatpiHQ You think it's doable (regarding my last message)? I can't seem to figure the expression for WiCan Pro that would do the calculation itself and send processed data.