meatpiHQ/wican-fw #507: Go to silent CAN mode when WiFi station is not available

StefanR71, 2025-08-06:
Hello,

is it possible to switch automatically to the silent CAN mode when the WiFi station isn't connected?

Problem:

When  i lock my Cupra Tavascan and the WiCAN module is still working on the bus, this triggers the alarm of the car. So i need to pull the WiCAN when i want to  lock the car.

When i'm at home and charge at my local charging station, i want to use the WiCAN to watch the SoC, here the car is in a garage which is closed, i don't lock my car here, then the WiCAN can be active.

Kind regards
Stefan

StefanR71, 2025-10-30:
Can't disable that in the car, because it is not possible in the 2025 series.

arcidodo, 2025-10-30:
sorry this is SFD not SFD2 my bad :-( i think you car has SFD2 Protection.

StefanR71, 2025-11-03:
Has something changed? I installed the 4.13 firmware a few days ago to made some tests...

Yesterday i had forgotten to pull the device, but the alarm of the car wasn't triggered.

Possible that the entry's in Automate don't be polled if the Mosquitto isn't connected or something that kind? I havn't tryed to lock the car at home, this i can test tomorrow.

alistair23, 2025-11-08:
Have a look at https://github.com/meatpiHQ/wican-fw/pull/592 for the WiCAN Pro.

Some in vehicle testing would be great, I'm not sure if it's fast enough or reliable enough to avoid the alarms

StefanR71, 2025-11-08:
Sorry, i don't have an pro...