meatpiHQ/wican-fw #511: New vehicule request : Opel Astra electric

TheRedstarr, 2025-08-11:
Hello,
I have connected my WICAN-USB to my car and I am able to read the SoC in Carscanner with ELM327 protocol (among others things).

I would like to be able to read the SoC in Home Assistant :
I successfully integrated Wican in it and configured the device with AutoPid but the Opel or Vauxhall Astra Electric is not listed in the vehicule specific Models. I have tried several others of Stellantis but in HA I have only few basics like Battery voltage or ECU Status.
Instructions says to open an issue here and I would be very happy to follow your instructions to add the vehicule.
Thank you in advance.
Louis

TheRedstarr, 2025-08-11:
Add-on : 
Standard PIDs does not work.
6- ISO 15765-4 (CAN 11bit/500K) returns "PID scan failed. NetworkError when attempting to fetch resource" despite this is the one marked as used by carscanner.
Other protocols return "No PIDs found. Try a different protocol or check if ignition is ON."

Then I tried to observe the log file of carscanner to see how it does and with inspiration of another vehicule model I set the following:
Specific Initialization : ATZ;ATE0;ATH1;ATSP6;ATS0;ATM0;ATAT1;ATSH6B4;ATFCSH6B4;ATFCSD300000;ATFCSM1;ATCRA694;

I set a SoC PID 22D410 with init ATSH6B4;ATFCSH6B4;ATCRA694; and [B5:B6]/512
It returns SOC	56.64 on wican_xxx.local/autopid_data which match exactly the figure shown by carscanner and is a little bit under the figure shown on the dash of the car, but I believe it is normal as the SoC is the % of the brut capacity of the battery and the dash the % of the net capacity of the battery which is lower.

Another PID 22D860 for the SoH with init ATSH6B4;ATFCSH6B4;ATCRA694; and [B5:B6]/16 retuns the value shown by carscanner.

Maybe it can help to set up a new car profile.

TheRedstarr, 2025-08-15:
Here is the file with several minutes of just the SoC, which varied between 75,58 % to 75,5 %

[log.txt](https://github.com/user-attachments/files/21797141/log.txt)

Formstation, 2025-10-22:
Thanks a lot for your quick response. Next challenge is, to get the data into evcc. There I found the info, that I need a modified firmware for WICAN dongle to get the right data via MQTT... I guess this will take some time...

Formstation, 2025-10-30:
@MatthiasGuelck I can only answer this question to the extent that I've done it so far. I'm receiving a SoC value in MQTT, but I'm not yet able to use it in evcc. I've asked for help elsewhere about this, but haven't received a reply yet. 

[github.com/meatpiHQ/wican-fw/discussions/278](https://github.com/meatpiHQ/wican-fw/discussions/278#discussioncomment-14787300)

TheRedstarr, 2025-10-30:
@meatpiHQ sorry I cannot tell you if your new profile works because I still have the custom working one I set initially. Mainly because I do not have time to make the change and validate it still work.
I recognize from the discussion that both @Formstation and @MatthiasGuelck get the soc in MQTT with your new profile, then you can probably close the case. Many thanks for your work.

@Formstation and @MatthiasGuelck : I cannot really help you with evcc as I do not use it, but I can tell you that in Home Assistant I had to manage the problem of the OBD of the Astra that shutdown when we stop the car, meaning that we cannot monitor the charge but only estimate it from the quantity of current sent. Maybe you have to find a way in evcc to manage this also.

MatthiasGuelck, 2025-11-03:
I use the WiCan adapter to transfer the SoC of my Opel Astra to an openWB. This works perfectly thanks to the detailed description in the openWB wiki.
The only thing I needed to clarify was which PIDs I had to use. But this is already correctly entered in the current version of the WiCan software.
Unfortunately, I still have no idea how to display the SoH.

MatthiasGuelck, 2025-11-03:
@Formstation:
Thank you for trying to help me.
Unfortunately, I didn't understand your advice.
Where/how should I change mosquitto.conf?

Formstation, 2025-11-03:
@MatthiasGuelck I don't know on which system you're running mqtt. There you should search the filesystem for "mosquitto.conf". Open it in any text-editor and add this lines, press "safe" and restart mqtt-server. 

@meatpiHQ Maybe it has something to do with the capacity of the Opel-Battery? It actually is 54 kWh but you just can use 51 kWh. I would try to change the value of the battery size. Just a guess though.

Formstation, 2025-11-03:
@MatthiasGuelck Maybe you try to search for "openWB mqtt config" in google? Sorry, I don't use either OpenWB or HA.