meatpiHQ/wican-fw #512: AutoPID stops responding after some time

Brazen00, 2025-08-12:
I've purchased a WiCAN and have it plugged into my Ioniq5, feeding data to home assistant via the official integration.

Initially, the WiCAN and HA communicate and HA displays the current values of the PIDs. However after a while, the values in HA remain static and the ECU Status diagnostic switches to "off". When I then access the WiCAN via the web portal and navigate to <ip address>/autopid_data, the values there match the most recent values in HA, and they never change again. I can kick-start it by rebooting the WiCAN, although this doesn't always work.

I can access the WiCAN web portal at all times, so we are not having a connectivity issue.

I've attached my config in case I've got something set up wrong.

[image]

[config_2025-08-12.json](https://github.com/user-attachments/files/21726259/config_2025-08-12.json)

meatpiHQ, 2025-08-12:
@Brazen00 Please try this firmware and see if it helps

[wican-fw_obd_v412-63-gc8d257c.zip](https://github.com/user-attachments/files/21738576/wican-fw_obd_v412-63-gc8d257c.zip)

Brazen00, 2025-08-13:
No, that hasn't helped. After loading the firmware I can still connect to the web portal, but I haven't been able to get any data from /autopid_data or from HA. Rebooting has not kicked it into gear at all. Additionally, HA is now reporting all sensors as Unavailable.

Brazen00, 2025-08-19:
@meatpiHQ do you have any further suggestions for this issue?

meatpiHQ, 2025-08-26:
@Brazen00 replacing the device will not fix this issue. It is possible that the ECU is going to sleep. I can send you a firmware with debug logs enabled this would us find the issue.

Brazen00, 2025-08-28:
here is a log just after reboot, where the reboot successfully updated the autopid values:

[wican_logs_20250828_101813.txt](https://github.com/user-attachments/files/22015368/wican_logs_20250828_101813.txt)

meatpiHQ, 2025-08-30:
@Brazen00 it seems like the ECU is going to sleep, is the car connected to the charger when this happens?

The odometer PID seems not working, i think you should remove it.

Brazen00, 2025-08-30:
Yes this happens when the car is charging, not charging, on, off, no matter the car state.

meatpiHQ, 2025-08-30:
@Brazen00 please try to remove the Odometer and try again

Brazen00, 2025-08-30:
I’ve removed Odometer and HV_V , as these were both failing. Initial signs are that the PIDs are updating as expected. So far so good!

meatpiHQ, 2025-08-30:
@Brazen00 keep me updated please. What's the vehicle manufacturering year?

domectrl, 2025-10-23:
Exactly same issue here. Ioniq 5 from 2021. Removing odometer and HV_V fixes the issue.

meatpiHQ, 2025-10-23:
@domectrl Are you running the latest firmware version?