meatpiHQ/wican-fw #514: AutoPID: Incorrect byte offset for ISO-TP multi frame responses

philipkocanda, 2025-08-14:
I’ve noticed it’s really challenging to add new PIDs to the AutoPID feature. I have dozens of responses logged from using Car Scanner and SavvyCAN (ISO-TO decoder), and the byte offsets just don’t line up with what WiCAN expects.

See also: https://github.com/meatpiHQ/wican-fw/issues/478#issuecomment-3182467692

After spending days on this issue, I think I know what’s going on. The WiCAN byte indexes do not map to the actual position in the [ISO-TP](https://en.wikipedia.org/wiki/ISO_15765-2#List_of_protocol_control_information_field_types) payload of the UDS response. Example ISO-TP Payload (obtained using SavvyCAN and its ISO-TP decoder):

```
62 B0 02 E0 00 00 00 00 AB 00 FE 07 00 00 00
                           ^^ ^^ ^^
                           09 10 11                       
```

Notice the byte index (zero-based) should be 9-11, but for WiCAN we need to tell it to look at bytes 12-14. 

**Here's why:** WiCAN includes the first two bytes from each ISO-TP Consecutive Frame, which are not supposed to be part of the payload as far as I understand. Car Scanner and Torque also skip those. This leads me to suspect WiCAN isn't reassembling the ISO-TP multi-frame response payload, but is just concatenating the received CAN messages, and basing its byte index off of that. 

This may also lead to data corruption if frames are received out of order, which I've actually observed when using multiple different polling intervals simultaneously. The ideal solution here would be to implement ISO-TP multi-frame support and re-assemble the payload accordingly. 

In the meantime, I’ve worked out a conversion table using this little python script:

```python
def payload_index_to_wican(ipa):
    """
    Convert zero-based index in ISO-TP payload (Ipa) to 
    zero-based index in frame (Ifr).
    
    Args:
        ipa (int): Zero-based index in ISO-TP payload
        
    Returns:
        int: Zero-based index in frame
    """
    if ipa < 6:
        # First frame: 2-byte ISO-TP header + payload
        return ipa + 2
    else:
        # Subsequent frames: 1-byte continuation header + payload
        remaining_bytes = ipa - 6
        frame_number = 1 + (remaining_bytes // 7)
        position_in_frame = 1 + (remaining_bytes % 7)
        frame_start_index = frame_number * 8
        return frame_start_index + position_in_frame

# Test cases from your data
print(f"BA: Ipa=6 -> Ifr={payload_index_to_wican(6)}")   # Should be 9
print(f"0F: Ipa=14 -> Ifr={payload_index_to_wican(14)}") # Should be 18

# Output:
# BA: Ipa=6 -> Ifr=9
# 0F: Ipa=14 -> Ifr=18
```

And a table showing the offsets (up to 60 bytes payload size):

| ISO-TP Payload Index | WiCAN Index | Frame | Pos in Frame | Frame Type |
| --------- | --------- | ----- | ------------ | ---------- |
| **0**     | 2         | 0     | 2            | FF         |
| **1**     | 3         | 0     | 3            | FF         |
| **2**     | 4         | 0     | 4            | FF         |
| **3**     | 5         | 0     | 5            | FF         |
| **4**     | 6         | 0     | 6            | FF         |
| **5**     | 7         | 0     | 7            | FF         |
| **6**     | **9**     | 1     | 1            | CF 1       |
| **7**     | 10        | 1     | 2            | CF 1       |
| **8**     | 11        | 1     | 3            | CF 1       |
| **9**     | 12        | 1     | 4            | CF 1       |
| **10**    | 13        | 1     | 5            | CF 1       |
| **11**    | 14        | 1     | 6            | CF 1       |
| **12**    | 15        | 1     | 7            | CF 1       |
| **13**    | 17        | 2     | 1            | CF 2       |
| **14**    | **18**    | 2     | 2            | CF 2       |
| **15**    | 19        | 2     | 3            | CF 2       |
| **16**    | 20        | 2     | 4            | CF 2       |
| **17**    | 21        | 2     | 5            | CF 2       |
| **18**    | 22        | 2     | 6            | CF 2       |
| **19**    | 23        | 2     | 7            | CF 2       |
| **20**    | 25        | 3     | 1            | CF 3       |
| **21**    | 26        | 3     | 2            | CF 3       |
| **22**    | 27        | 3     | 3            | CF 3       |
| **23**    | 28        | 3     | 4            | CF 3       |
| **24**    | 29        | 3     | 5            | CF 3       |
| **25**    | 30        | 3     | 6            | CF 3       |
| **26**    | 31        | 3     | 7            | CF 3       |
| **27**    | 33        | 4     | 1            | CF 4       |
| **28**    | 34        | 4     | 2            | CF 4       |
| **29**    | 35        | 4     | 3            | CF 4       |
| **30**    | 36        | 4     | 4            | CF 4       |
| **31**    | 37        | 4     | 5            | CF 4       |
| **32**    | 38        | 4     | 6            | CF 4       |
| **33**    | 39        | 4     | 7            | CF 4       |
| **34**    | 41        | 5     | 1            | CF 5       |
| **35**    | 42        | 5     | 2            | CF 5       |
| **36**    | 43        | 5     | 3            | CF 5       |
| **37**    | 44        | 5     | 4            | CF 5       |
| **38**    | 45        | 5     | 5            | CF 5       |
| **39**    | 46        | 5     | 6            | CF 5       |
| **40**    | 47        | 5     | 7            | CF 5       |
| **41**    | 49        | 6     | 1            | CF 6       |
| **42**    | 50        | 6     | 2            | CF 6       |
| **43**    | 51        | 6     | 3            | CF 6       |
| **44**    | 52        | 6     | 4            | CF 6       |
| **45**    | 53        | 6     | 5            | CF 6       |
| **46**    | 54        | 6     | 6            | CF 6       |
| **47**    | 55        | 6     | 7            | CF 6       |
| **48**    | 57        | 7     | 1            | CF 7       |
| **49**    | 58        | 7     | 2            | CF 7       |
| **50**    | 59        | 7     | 3            | CF 7       |
| **51**    | 60        | 7     | 4            | CF 7       |
| **52**    | 61        | 7     | 5            | CF 7       |
| **53**    | 62        | 7     | 6            | CF 7       |
| **54**    | 63        | 7     | 7            | CF 7       |
| **55**    | 65        | 8     | 1            | CF 8       |
| **56**    | 66        | 8     | 2            | CF 8       |
| **57**    | 67        | 8     | 3            | CF 8       |
| **58**    | 68        | 8     | 4            | CF 8       |
| **59**    | 69        | 8     | 5            | CF 8       |
| **60**    | 70        | 8     | 6            | CF 8       |

[Frame Types](https://en.wikipedia.org/wiki/ISO_15765-2#List_of_protocol_control_information_field_types):
  FF = First Frame
  CF = Continuation frame

meatpiHQ, 2025-08-14:
@philipkocanda Thanks for that! The flow control is not properly implemented I think. I can try to fix, it would you able to reproduce and test or somehow have a python script that simulates the issue?

philipkocanda, 2025-08-14:
I can give you the raw CAN frames of the response, would that help?

philipkocanda, 2025-08-14:
Raw CAN response frames:

```
7EC103D6101FFFFFFFF
7EC21BA1D2E26480300
7EC22180F4714131313
7EC231414130015CB4D
7EC24CB010000860004
7EC25C9B70004C80A00
7EC2601BB780001ADDB
7EC270125A6B70D0186
7EC280000000003E800
```

Imported in SavvyCAN:

[image]

SavvyCAN's ISO-TP decoder:

[image]

Decoded ISO-TP payload:

```
61 01 FF FF FF FF BA 1D 2E 26 48 03 00 18 0F 47 14 13 13 13 14 14 13 00 15 CB 4D CB 01 00 00 86 00 04 C9 B7 00 04 C8 0A 00 01 BB 78 00 01 AD DB 01 25 A6 B7 0D 01 86 00 00 00 00 03 E8 00 00
```

Payload decoded in Hex editor (ImHex):

[image]

meatpiHQ, 2025-08-14:
I need an example of a bad response as well

philipkocanda, 2025-08-15:
> I need an example of a bad response as well

By bad response, do you mean the flow control issue (messages getting mixed up) or do you mean the byte offsets? 

In case of the flow control issue, the issue happened when I had multiple PIDs being requested at *different* intervals (2500ms, 30000ms, 5000ms). I'll share the JSON export of my settings and a screenshot of the corrupted data later today.

Regarding the byte offsets, from what I can see in my Car Scanner logs, ELM327 has no opinion on the payload of UDS/ISO-TP responses, it just returns the raw CAN frames.

It's up to the application consuming them to reassemble the payload. WiCAN interprets the payload like this (full CAN frames):

```
10 3D 61 01 FF FF FF FF
21 BA 1D 2E 26 48 03 00
22 18 0F 47 14 13 13 13
23 14 14 13 00 15 CB 4D
24 CB 01 00 00 86 00 04
25 C9 B7 00 04 C8 0A 00
26 01 BB 78 00 01 AD DB
27 01 25 A6 B7 0D 01 86
28 00 00 00 00 03 E8 00
```

While the payload according to UDS/ISO-TP is (excluding the PCI byte(s)):

```
61 01 FF FF FF FF
BA 1D 2E 26 48 03 00
18 0F 47 14 13 13 13
14 14 13 00 15 CB 4D
CB 01 00 00 86 00 04
C9 B7 00 04 C8 0A 00
01 BB 78 00 01 AD DB
01 25 A6 B7 0D 01 86
00 00 00 00 03 E8 00
```

This is where the problem of incorrect byte offsets happens. Note the ISO-TP PCI byte(s) are included in what WiCAN considers to be the payload. It also requires WiCAN users have to have understanding of a transport protocol (ISO-TP) underneath the application layer, which is rather confusing and not something other applications require you to do.

Another diagram clarifying the distinction between payload (data, blue) and protocol information (gray) for each ISO-TP frame type:

[image]

philipkocanda, 2025-08-15:
Another reason why this is a problem - rather than being able to use the byte range notation for a UINT32 value (`[B32:B35]`), I had to resort to manually adding up the individual bytes due to the PCI byte being **right in the middle of the UINT32 value I wanted to address**:

```json
{
  "pid": "2101",
  "pid_init": "",
  "parameters": [
    {
      "name": "CUMULATIVE_ENERGY_CHARGED",
      "expression": "(B47<<24)|(B49<<16)|(B50<<8)|B51",
      "unit": "",
      "class": "",
      "period": "5000",
      "min": "",
      "max": "",
      "type": "Default",
      "send_to": ""
    }
  ]
},
```

meatpiHQ, 2025-08-19:
> Out of curiosity, what makes you think it’s not a good idea? Is it (lack of) backwards compatibility?

Yes backward compatibility is a big issue, you won't be able to use any app like car scanner or torque. 

You can read more about elm327 flow control in the datasheet.

https://cdn.sparkfun.com/assets/learn_tutorials/8/3/ELM327DS.pdf

philipkocanda, 2025-08-21:
> Yes backward compatibility is a big issue, you won't be able to use any app like car scanner or torque.

That doesn't seem like it should be mutually exclusive?

Technically it should be possible to expose the raw CAN frames in ELM327 mode, and have another mode (AutoPID2?) which supports ISO-TP payloads.